PowerBroker2/SerialTransfer #96: File Transfer Not wokring as Expected

amals, 2022-10-20:
**Describe the bug**
I am trying to capture Image from ESP-CAM and send it over serial transfer to another ESP32. 
I have got over here through a Forum where PowerBroker2 explained very clearly and got it working for the user it seems. But the version is outdated and I am trying to get it using latest version. However even the example code posted here is not working as expected. Its not even compiling. I have to change the “Filesize” int to 2001 to make it compile. And from the receiving end i got only 1442 characters all the time. 

**To Reproduce**
I have used the example file uart_rx_file.ino and uart_tx_file.ino

**Expected behavior**
It should transfer all the data at least once.

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser 
 - Version 3.1.3

**Additional context**
I have more queries to ask but not in bug report. If its fixed I can create new thread.

PowerBroker2, 2022-10-20:
Are you getting any transfer/parsing errors? These would cause dropped packets.

amals, 2022-11-05:
> Are you getting any transfer/parsing errors? These would cause dropped packets.

Hi Sorry for late reply. Was busy on project delivery. Given up serial transfer and taken different route(ESPNOW Transfer). On Serial Transfer, no Errors reported. Just the data is missing. Probably we will work next week to trace the issue again. Please direct me how should I begin to resolve this bug? Did you try yourself with the example code you provided?

PowerBroker2, 2022-11-05:
> Did you try yourself with the example code you provided?

It's been a while since I tested it, but I remember doing so and it worked then

> Please direct me how should I begin to resolve this bug?

I don't have any ideas apart from asking on the Arduino or Espressif forums

closed 2025-07-14 by PowerBroker2