PowerBroker2/SerialTransfer #97: Received data always 7 (after converted to int)

askiiart, 2022-10-24:
My code receives an int from 0 to 100 from my PC, converts it to an int, and displays the int in binary using some LED lights. However, the lights always display the number 7.

Code is below:
```
void loop() {
  if(myTransfer.available())
  {

    // send all received data back to Python
    for(uint16_t i=0; i < myTransfer.bytesRead; i++){
      myTransfer.packet.txBuff[i] = myTransfer.packet.rxBuff[i];
    }
    myTransfer.sendData(myTransfer.bytesRead);

    int x = (int) myTransfer.packet.rxBuff;
    if((x % 2) > 0) { digitalWrite(pin1, HIGH); } else { digitalWrite(pin1, LOW); }
    if((x % 4) > 1) { digitalWrite(pin2, HIGH); } else { digitalWrite(pin2, LOW); }
    if((x % 8) > 3) { digitalWrite(pin3, HIGH); } else { digitalWrite(pin3, LOW); }
    if((x % 16) > 7) { digitalWrite(pin4, HIGH); } else { digitalWrite(pin4, LOW); }
    if((x % 32) > 15) { digitalWrite(pin4, HIGH); } else { digitalWrite(pin4, LOW); }
    if((x % 64) > 31) { digitalWrite(pin4, HIGH); } else { digitalWrite(pin4, LOW); }
    if((x % 128) > 63) { digitalWrite(pin4, HIGH); } else { digitalWrite(pin4, LOW); }
  }
}
```

PowerBroker2, 2022-10-24:
Try using `&` instead of `%`

askiiart, 2022-10-24:
> Try using `&` instead of `%`

I just got home to test that; that makes it display 15

PowerBroker2, 2022-10-24:
Oh wait, why are you doing this?:

```C++
int x = (int) myTransfer.packet.rxBuff;
```

That typecasts the `rxBuff` pointer to an int and doesn't do anything with the data inside. I suggest looking at the [UART TX](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/uart_tx_data/uart_tx_data.ino) and [UART RX](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/uart_rx_data/uart_rx_data.ino) examples for better usage of the library.

askiiart, 2022-10-25:
I made a struct before this code, and added:
```
uint16_t recSize = 0;
    recSize = myTransfer.rxObj(myStruct, recSize);
    int x = myStruct.val * 2;
```
It's now working, thanks for the help! I don't understand why I needed to multiply `myStruct.val` by 2, though. Would you mind explaining that to me?

closed 2022-10-25 by askiiart

PowerBroker2, 2022-10-25:
I can't tell without seeing your code, but as an aside: no need to include `recSize`. It's in the wrong place within the function parameter list and isn't needed anyway (as far as I can tell)