PowerBroker2/SerialTransfer #102: Why is the main code commented?

RobertLRead, 2022-11-30:
With all due respect, I'm deviating from the bug format---this is just a question.

Why is all the functionality in SPITransfer.h and SPITransfer.cpp commented out?

I have uncommented it, and can compile. Perhaps this is a clever way to make someone read the code. I am sure this is not a mistake, it must have some purpose, but I can't fathom it.

PowerBroker2, 2022-11-30:
Unfortunately not all Arduinos/Arduino compatible MCUs have compatible SPI APIs. I developed the original SPI portion of the library with the standard API, but it breaks for others using different boards even when they aren't using the SPI portion. I decided to stop supporting SPI for this lib, but left the source code available via the commented files in case someone really wanted to use it.

There are several issues about it, but this one comes to mind first: https://github.com/PowerBroker2/SerialTransfer/issues/34#issuecomment-662712105

RobertLRead, 2022-11-30:
Thank you, that makes sense. Closing this issue.

closed 2022-11-30 by RobertLRead