PowerBroker2/SerialTransfer #103: Use this to send a file update from device A to B

RosiersRobin, 2022-12-06:
Hi,

I've tried to use this package to update an second ESP32, atached to the UART1 (Serial1).

So my use-case is the following:

- 2 ESP's, connected to eachother via UART1.
- 1 has internet using a LAN-chip and thus connected via wire. The other one is not Wi-Fi capable, so I can't update using wifi, since the device can change location any time.

I want to download the firmware for te second, non internet connected eps32, onto the ethernet connected one. This works as intended.

After that download, I want to send/stream the downloaed file to the other ESP, that does not have internet and after the file got streamed, flash it to the ESP.

What I currently have:

**Sender**
```cpp
File readFile = LittleFS.open("/newFirmware.bin");

uint8_t buf[1025] = {};
size_t _len = 256;

serialTransferClient.begin(Serial1);

serialTransferClient.sendDatum("newFirmware.bin"); // Send filename

uint16_t fileSize = readFile.size();

uint16_t numPackets = fileSize / (MAX_PACKET_SIZE - 2); // Reserve two bytes for current file index

// Add an extra transmission if needed
if (fileSize % MAX_PACKET_SIZE) {
       numPackets++;
}

// Send all data within the file across multiple packets
for (uint16_t i = 0; i < numPackets; i++) {

    int len = readFile.read(buf, _len);

    uint16_t fileIndex = i * MAX_PACKET_SIZE; // Determine the current file index
    uint8_t dataLen = MAX_PACKET_SIZE - 2;

    // Determine data length for the last packet if file length is not an exact multiple of MAX_PACKET_SIZE-2
    if ((fileIndex + (MAX_PACKET_SIZE - 2)) > fileSize) {
        dataLen = fileSize - fileIndex;
    }

    uint8_t sendSize = serialTransferClient.txObj(fileIndex); // Stuff the current file index
    sendSize = serialTransferClient.txObj(buf[fileIndex], sendSize, dataLen); // Stuff the current file data

    serialTransferClient.sendData(sendSize, 1); // Send the current file index and data
    delay(100);
}
```

**Receiver:**
```cpp

serialTransferClient.begin(Serial1);

while (serialTransferClient.available() && !finished) {
    if (serialTransferClient.available()) {
        if (!serialTransferClient.currentPacketID()) {
            serialTransferClient.rxObj("newFirmware.bin");
        } else if (serialTransferClient.currentPacketID() == 1) {
            for (uint8_t i = 2; i < serialTransferClient.bytesRead; i++) {
                // Serial.print((char)serialTransferClient.packet.rxBuff[i]);
                Update.write(serialTransferClient.packet.rxBuff[i], l); // this does not work either...
            }
        }
    }

    lastTime = millis();
}
```

Any help would be really appriciated!

cyberdevnet, 2023-10-29:
@RosiersRobin did you manage to update one esp32 from another? It's exactly what I'm trying to do...not too much info over the internet

RosiersRobin, 2023-10-31:
@cyberdevnet I did, however not by using this package, but by transferring 64bytes each chunk. So I don't have to do specific calculations and buffering. There is room for optimizing this, but I haven't had the time yet.

nomakewan, 2025-04-25:
Looking at the original code, could just be an ESP32-specific thing? I know you said you resolved this, but to get to the original problem [someone mentioned](https://esp32.com/viewtopic.php?p=99160&sid=1f0c18e8922069f62a97266a45be5fb8#p99160) that `Update.h` expects the buffer size to be `4096`. This isn't a huge problem on your TX side because you're not actually running `Update` there, so it's irrelevant that you have it set to `1025`. But on the RX side, you're calling the `serialTransferClient.packet.rxBuff` which according to the source code of this library is `rxBuff[MAX_PACKET_SIZE]` which translates to `rxBuff[254]`, far below `4096`.

So my thought would be to have a separate 4096-byte-wide buffer for the update operation, which is then filled by the RX buffer on the fly and once it contains 4096 bytes, that chunk is passed to `Update.write`. Of course you would need to make sure you do not exit the update while loop prematurely, so would also want the transmit side to send a piece of data indicating that the transfer of the binary file is complete and trigger on that to call `Update.end()`.

closed 2025-07-14 by PowerBroker2