PowerBroker2/SerialTransfer #106: Two way communication

fif911, 2023-01-22:
I am new to this library. But I have not found a way to achieve to way communication between Master and Slave. Is it possible with the current implementation?

PowerBroker2, 2023-01-22:
Yes, there are I2C examples in the [examples folder](https://github.com/PowerBroker2/SerialTransfer/tree/master/examples)

fif911, 2023-01-28:
I looked through them, but I cannot find a 2-way communication example. I was able to find only Master to Slave but not Slave to Master (can be done with requestFrom in wire lib). Is there a way to do smth like this? @PowerBroker2

Drikteur, 2023-05-04:
Also looking for 2 way communication on uart, tried to merge the sketches (TX and RX)and it works both ways but not at the same time, the arduino which boots latest acts as master... (trying all day but to stupid to make it work :-))

closed 2025-07-14 by PowerBroker2