PowerBroker2/SerialTransfer #108: Data Transfer Between ESP 32 And Arduino Mega

mostafaemad10, 2023-03-29:
I want to transfer data from arduino mega to Esp32 when I use the example sketches ( uart_tx_data on arduino and uart_rx_data on esp32 ) the code generates wrong output ($ovf | ld) . i tried many options i found out that when the whole structure is of the same type (float/bool) it works well but when i mix between data types in the same structure it generates wrong output depends on the order of structure elements (if the first element is char , second float and third is float the output becomes the char value then  all 0 but if the order is float char float the output becomes first value is right then char is right then the third element is 0  so the order matters ) also if the data type is int or double the output becomes totally wrong values. 

I need to send data from the arduino mega to the Esp32 in a structure with mixed data types if possible and i need in these types double ( 8 bytes ). 
Is this possible and if you could help with any tips or recommendation.
Thanks in advance.

PowerBroker2, 2023-03-29:
This is really interesting - the examples should work for arduino <-> ESP32. How about revamping the examples so that you're transferring dummy data that matches what you want to transfer in your project? We can debug further from there for your specific case.

proasnet, 2024-07-19:
Dear @PowerBroker2, I am trying the same. ESP32 as reader and Atmel as sender. Compilling both, ok.
But ESP32 cyclic restarting.

ESP32 code
```
#include "SerialTransfer.h"
#include <HardwareSerial.h>

#define RX 14
#define TX 12

SerialTransfer myTransfer;

struct __attribute__((packed)) STRUCT {
  char z;
  float y;
} testStruct;

void setup() {
  Serial2.begin( 115200 , SERIAL_8N1 , RX , TX , false );
}

void loop() {
    if(myTransfer.available())
  {
    myTransfer.rxObj(testStruct);
    Serial.print(testStruct.z);
    Serial.println(testStruct.y);
  }
}
```

Atmel code
```
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct __attribute__((packed)) STRUCT {
  char z;
  float y;
} testStruct;

void setup() {
  Serial1.begin( 115200 );
  testStruct.z = '$';
  testStruct.y = 4.5;
}

void loop() {
  myTransfer.sendDatum(testStruct);
  delay(500);
}

esp console, repeating cyclic
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d1738  PS      : 0x00060830  A0      : 0x800d1418  A1      : 0x3ffb2240  
A2      : 0x3ffc1c58  A3      : 0xffffffff  A4      : 0x0000000e  A5      : 0x0000000e  
A6      : 0x0000000c  A7      : 0x00000078  A8      : 0x800d2084  A9      : 0x3ffb21d0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffb7d48  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40085f6c  LEND    : 0x40085f77  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1735:0x3ffb2240 0x400d1415:0x3ffb2270 0x400d26b1:0x3ffb2290

ELF file SHA256: 9ca8412a952522ad

Rebooting...
```

```

nomakewan, 2024-09-22:
> Dear @PowerBroker2, I am trying the same. ESP32 as reader and Atmel as sender. Compilling both, ok. But ESP32 cyclic restarting.

You forgot to initialize your SerialTransfer object on the Serial port you intend to use for inter-device communication. So for the ESP32, after the line in setup() `Serial2.begin( 115200 , SERIAL_8N1 , RX , TX , false );` you would put `myTransfer.begin(Serial2);`. Then on the Atmel, after the line in setup() `Serial1.begin( 115200 );` you would put `myTransfer.begin(Serial1);`.

For OP, I know it's been over a year since you posted and you never responded to the maintainer, but ESP32<->Mega with arbitrary data types works perfectly fine. You simply use a struct to store the variables and their types, and make sure that that struct's definition is identical on both ends. As long as both ends have the same definition for the struct, it will be correctly interpreted on the receiving end. A project I work on has two Mega 2560s and an ESP32 and they all use SerialTransfer to send multiple arbitrary data types. Works a treat!

proasnet, 2024-09-23:
Thank you @PowerBroker2

nomakewan, 2025-04-25:
For OP's problem, this is why you should never use a datatype like `char` or `int` when doing serial data between vastly different devices. Instead you should use strict definitions, so instead of `char` use `uint8_t`.

The issue here is that on the Mega 2560, `char` is the same as `int8_t`. But on the ESP32, `char` is `uint8_t`. Why does ESP32 do this? Who knows. But it doesn't matter, because this is a very important lesson about code portability. You cannot be sure that a data type like `char` or `word` will behave the same on every device, because history shows that this is a dangerous assumption.

That lesson out of the way, and seeing as OP has not returned, it is probably safe to close this issue. :)

closed 2025-07-14 by PowerBroker2