PowerBroker2/SerialTransfer #109: Documentation on how to send&recieve a single datum struct object (via serial) that exceeds the 254-byte maximum packet size

davidsandbrand, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
The examples all presume an object is under the 254-byte limit.

**Describe the solution you'd like**
An example.

**Describe alternatives you've considered**
I have tried to adapt the code in the 'uart_tx_file' & 'uart_rx_file' examples, but cannot seem to figure out how to recombine the chunks

**Additional context**
I have a struct that contains 94 float values and (obviously) only the first 63 make it through.  Unfortunately, I cannot debug via the serial port as I'm using a single board with both an AT328 & ESP8266 and they connect via the hardware serial port, so I have **_either_** a serial connection to monitor, **_or_** a connection between the two chips - never both, making this very hard to diagnose.  Any help would be appreciated.

Thank you!

davidsandbrand, 2023-04-17:
this is the board, FYI in case it's relevant: https://www.hackster.io/umpheki/uno-r3-wifi-esp8266-ch340g-arduino-and-wifi-single-board-eed9f6

nomakewan, 2025-04-25:
Yeah, the problem is that the current file example only works because of a technicality; the transmit is fine, but the receive end just spews anything it receives out the serial terminal (except for the filename, which works because the first packet is ID 0, which when converted to bool is `false`).

I honestly believe this is a really good suggestion. The trouble with implementing it is that while the transmit end will always know how big any arbitrary file is (after all, it is the transmitter), the receive end may not necessarily know how much memory to allocate in order to store the file. On a modern micro this isn't a huge deal, we have things like vectors for that. Arduino however doesn't natively have vectors so you would need to dynamically allocate a potentially massive array which has problems of its own.

If you control both ends of the communication however, of course you can just preallocate sufficient array space. In fact the existing file RX example does this despite not using it, so I might just rewrite it to properly print the array contents rather than spitting out characters piecemeal.