PowerBroker2/SerialTransfer #110: Compile time error on STM32F103 BluePill

cmr772023, 2023-05-16:
**Describe the bug**
The simplest test program fails to compile giving the following error:

Non-verbose compiler output:
Arduino: 1.8.13 (Windows 10), TD: 1.56, Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STM32duino bootloader, 72Mhz (Normal), Smallest (default)"
C:\Users\cmr\Documents\Arduino\libraries\SerialTransfer\src\I2CTransfer.cpp: In member function 'void I2CTransfer::begin(TwoWire&, const configST&)':
C:\Users\cmr\Documents\Arduino\libraries\SerialTransfer\src\I2CTransfer.cpp:21:8: error: 'class TwoWire' has no member named 'onReceive'
  port->onReceive((void (*)(int))processData);
        ^
Multiple libraries were found for "Wire.h"
 Used: C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26\libraries\Wire
 Not used: C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26\libraries\WireSlave
exit status 1
Error compiling for board Generic STM32F103C series.

Verbose Compiler output:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cmr\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cmr\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cmr\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\cmr\AppData\Local\Temp\arduino_build_723837 -warnings=none -build-cache C:\Users\cmr\AppData\Local\Temp\arduino_cache_551173 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\cmr\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\cmr\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.stm32tools.path=C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26 -prefs=runtime.tools.stm32tools-2022.9.26.path=C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26 -verbose C:\Users\cmr\Desktop\Serial transfer Test\send\send.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cmr\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cmr\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cmr\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\cmr\AppData\Local\Temp\arduino_build_723837 -warnings=none -build-cache C:\Users\cmr\AppData\Local\Temp\arduino_cache_551173 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\cmr\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\cmr\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.stm32tools.path=C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26 -prefs=runtime.tools.stm32tools-2022.9.26.path=C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26 -verbose C:\Users\cmr\Desktop\Serial transfer Test\send\send.ino

Using board 'genericSTM32F103C' from platform in folder: C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26

Using core 'maple' from platform in folder: C:\Users\cmr\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26

Detecting libraries used...

"C:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/include" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/stm32f1/include" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/usb/stm32f1" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/usb/usb_lib" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\cores\\maple" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\variants\\generic_stm32f103c" "C:\\Users\\cmr\\AppData\\Local\\Temp\\arduino_build_723837\\sketch\\send.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for SerialTransfer.h: [SerialTransfer@3.1.3]

ResolveLibrary(SerialTransfer.h)

  -> candidates: [SerialTransfer@3.1.3]

"C:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/include" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/stm32f1/include" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/usb/stm32f1" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/usb/usb_lib" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\cores\\maple" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\variants\\generic_stm32f103c" "-IC:\\Users\\cmr\\Documents\\Arduino\\libraries\\SerialTransfer\\src" "C:\\Users\\cmr\\AppData\\Local\\Temp\\arduino_build_723837\\sketch\\send.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple" "-IC:\\Users\\cmr\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\system/libmaple/incl [... cut ...]
  testStruct.h = 55.6 + random(1, 15);
  Serial.println (testStruct.task);
  Serial.println (testStruct.t);
  Serial.println (testStruct.h);
  Serial.println("");
  
  myTransfer.sendDatum(testStruct);
  delay(1000);
}

**Expected behavior**

**Screenshots**
None

**Desktop (please complete the following information):**
 - Windows 10
 - Arduino IDE v1.8.13

**Additional context**
The same code compiles and runs on an ESP32 (only used the ESP32 as I needed two to test and more than one serial port to aid debugging

PowerBroker2, 2023-05-16:
STM32 must have a strange I2C API - try commenting out the I2C-related SerialTransfer files and includes. Should work after doing that

cmr772023, 2023-05-16:
Thanks for quick reply - I just removed the I2C & (for good measure as I'm only interested in serial) the SPI  .cpp & .h files - now compiles.
Thanks for your help.

cmr772023, 2023-05-17:
I thought that once it compiled I would be out of the woods but it appears not to be the case.
I've been attempting to get this to work between an Arduino Nano and an STM32F103 (Blue Pill) without success and was wondering if you have any ideas as to what the incompatibility might be.
Send Code - Arduino Nano:
//SerialTransfer.h test - send from Arduino Nano

#include "SerialTransfer.h"
SerialTransfer myTransfer;

struct TX {
  char TxTask;
  float T1;
  float T2;
  float T3;
} TxStruct;

void setup()
{
  Serial.begin(9600);
  myTransfer.begin(Serial);

  TxStruct.TxTask = 'A';
  TxStruct.T1 = 4.5 + random(1, 15);
  TxStruct.T2 = -5.6;
  TxStruct.T3 = 100.23;
}

void loop()
{
  myTransfer.sendDatum(TxStruct);
  delay(500);
}

Receive code - STM32F103
//SerialTransfer.h test - receive STM32F103

#include "SerialTransfer.h"
SerialTransfer myTransfer;

struct RX {
  char RxTask;
  float R1;
  float R2;
  float R3;
} RxStruct;

void setup()
{
  Serial.begin(115200);
  Serial2.begin(9600);
  myTransfer.begin(Serial2);
}

void loop()
{
  if (myTransfer.available())
  {
    myTransfer.rxObj(RxStruct);
    Serial.println(RxStruct.RxTask);
    Serial.println(RxStruct.R1);
    Serial.println(RxStruct.R2);
    Serial.println(RxStruct.R3);
    Serial.println("");
  }
  else if(myTransfer.status < 0)
  {
    Serial.print("ERROR: ");
    Serial.println(myTransfer.status);
  }
}

I'm sending  char + 3 Floats, what I'm receiving is:
A - that good as it's what is being sent
-0.00
-251663.0
0.00

All of the float data is incorrect.

I have tried between two STM32F103s and that does work. 
the Atmel and ST devices are both little endian so it's not that.
I've looked through the source but a fair bit of it is way over my head!

An oscilloscope reveals that the send data isn't changing (as the 2nd float should change: 4.5+random(1,15)) so I suspect somethings not right on the Nano.
Would you expect this to work between the two processor families or am expecting too much?

Carl

PowerBroker2, 2023-05-17:
Sounds like a question for the Arduino forum. Transfer should work between Arduinos of different type, however you might run into byte misalignment issues. These misalignment issues can be resolved using packed structs with explicit bit length types. From your o-scope probing, however, this may not be the issue

closed 2023-05-17 by PowerBroker2