PowerBroker2/SerialTransfer #113: This library assumes that it would be instantiated and never removed.

MartinMueller2003, 2024-03-29:
Good C++ Practice is to have a working class implement a constructor and a destructor. In this case, we have a begin, but no end function. There is no way to turn this off and remove the memory used after it is no longer needed.

**Describe the solution you'd like**
At the minimum implement an end function which releases all resources allocated by the class instance during operation. This will allow someone to "new" the class object and later "delete" it and not have a memory leak. A destructor would be preferred over an "End" but that would be the implementer's choice.

**Describe alternatives you've considered**
Still looking for a solution to have a web interface send a file to an Arduino. The memory for this would come from shutting down other functions while the transfer is in progress and then shut down the transfer function to regain the buffer space needed by other operations.

**Additional context**
Add any other context or screenshots about the feature request here.

PowerBroker2, 2024-03-29:
Thanks, this is a good idea. Yes, when I wrote it, it was intended for remote control robotics where you would have a continuous link throughout the entire lifetime of the sketch.

I don't have a lot of experience with destructors, so I'll have to do some research. If you're more experienced with destructors, you could take a stab at it - I'm always happy to review merge requests.

MartinMueller2003, 2024-03-29:
I forked the code and took a quick look. There is one structure that has a pointer to memory that is somehow filled in at run time (have not found where). Other than that, everything looks like it is static so adding a destructor that closes the Serial interfaces etc should be all that is needed. Yes I will add this and create a PR. 

For my project, the hard part will be writing the JavaScript to drive the PC side of the connection so that a browser application can generate the files being transfered. Using this to install an initial configuration for a device that had just been updated using the Web-tools java app.

PowerBroker2, 2024-03-29:
Which pointer are you referring to? I think there are a few different ones.

Also, are you not interested in FTP or is that not an option for your project?

MartinMueller2003, 2024-03-30:
It is not an option. It has to be via the serial port. I did consider SFTP over SSH but my application uses the serial ports as outputs in some of the implementations so I have to be able to turn off whatever I use. That means not using ssh. FYI: This is going to be used for the ESPixelStick V4 Web Based code update mechanism (Similar to the way WLED does the code updates.

closed 2026-11-06 by PowerBroker2