PowerBroker2/SerialTransfer #114: RS483 with SoftwareSerial not working as expected

MattiaIezzi, 2024-04-19:
Hi
I have been working on a dishwasher controller that needs two RS485 connected board to communicate. The slave board reads data from the buttons and displays data sent from the master. The master elaborates button inputs and sends data to display

My issue is that the display board does not send any data with the sendDatum() operation, and I can't seem to figure out why. I have tried basic communication between the two boards with just the sendDatum() and rxObj() but other than that it doesn't work. Can hopefully someone help me out? Below you'll find the master and slave codes, thanks

Master's code:

`#include "SerialTransfer.h"
#include <SoftwareSerial.h> // for software serial capability

const byte rxPin = 10;
const byte txPin = 11;
const byte enableTransmissionPin = 3;

SoftwareSerial swSerial =  SoftwareSerial(rxPin, txPin);
SerialTransfer RS485;

const byte BUTTONS = 8;

struct __attribute__((packed)) structSentData {

  byte hledstates; // prog, extradry, shine, hygiene, delay
  byte vledstates; // cycle running, end of cycle, salt missing, rinse aid missing
  char displayText[4]; // 4 due to null termination character
} dataToSend;

struct  __attribute__((packed)) structReceivedData {
  bool btnstates[BUTTONS]; // prog, extradry, shine, hygiene, autoclean, delay, start, cancel
} dataReceived;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  pinMode(rxPin, INPUT);
  pinMode(txPin, OUTPUT);
  pinMode(enableTransmissionPin, OUTPUT);

  swSerial.begin(9600);
  RS485.begin(swSerial);

  digitalWrite(enableTransmissionPin, LOW);
}

int counter = 0;

void loop() {
  // put your main code here, to run repeatedly:

  digitalWrite(enableTransmissionPin, LOW);

  if (RS485.available()) {
    Serial.print("a");
    RS485.rxObj(dataReceived);
  }

  digitalWrite(enableTransmissionPin, HIGH); // metti questo subito dopo la ricezione per evitare STALE_ERROR

  dataToSend.vledstates = B10100000;

  RS485.txObj(dataToSend, sizeof(dataToSend));
  RS485.sendData(sizeof(dataToSend));

  delay(50);

}`

Slave's code:

`/*
  Control panel code for Arduino controlled dishwasher

*/

const int shortPress = 50;
const int longPress = 3000;

const byte prog_btn_pin = 2;
const byte extra_dry_pin = 7;
const byte shine_plus_pin = 8;
const byte hygiene_pin = 9;
const byte delay_timer_pin = 12;
const byte start_pin = 13;

const byte rxPin = 10;
const byte txPin = 11;
const byte enableTransmissionPin = 3;
const byte latchPin = 5;
const byte dataPin = 6;
const byte clockPin = 4;

const byte BUTTONS = 8;

enum buttonsList {
  PROG,
  DRY,
  SHINE,
  HYGIENE,
  AUTOCLEAN,
  DELAY,
  START,
  CANCEL
};

#include "SerialTransfer.h"
#include <SoftwareSerial.h> // for software serial capability
#include <ShiftDisplay.h> // for 7 segment and leds control
#include <Debounce.h> // for button debouncing

Debounce prog_btn(prog_btn_pin, shortPress, true); // debounce the pin connected on pin corresponding to prog_btn_pin for 50 ms and consider it having internal pull_up
Debounce extradry_btn(extra_dry_pin, shortPress, true);
Debounce shineplus_btn(shine_plus_pin, shortPress, true);
Debounce hygiene_btn(hygiene_pin, shortPress, true);
Debounce autoclean_btn (hygiene_pin, longPress, true);
Debounce delaytimer_btn(delay_timer_pin, shortPress, true);
Debounce start_btn(start_pin, shortPress, true);
Debounce cancel_btn (start_pin, longPress, true);

ShiftDisplay display(latchPin, clockPin, dataPin, COMMON_ANODE, 5); // create the display object
SoftwareSerial swSerial =  SoftwareSerial(rxPin, txPin);
SerialTransfer RS485;

struct  __attribute__((packed)) structReceivedData {

  byte hledstates; // prog, extradry, shine, hygiene, delay
  byte vledstates; // cycle running, end of cycle, salt missing, rinse aid missing
  char displayText[4]; // 4 due to null termination character
} receivedData;

struct  __attribute__((packed)) structDataToSend {
  bool btnstates[BUTTONS] = {0, 0, 0, 0, 0, 0, 0, 0}; // prog, extradry, shine, hygiene, autoclean, delay, start, cancel
} dataToSend;

void setup() {

  pinMode(rxPin, INPUT);
  pinMode(txPin, OUTPUT);
  pinMode(enableTransmissionPin, OUTPUT);
  pinMode(latchPin, OUTPUT);
  pinMode(dataPin, OUTPUT);
  pinMode(clockPin, OUTPUT);

  pinMode(prog_btn_pin, INPUT_PULLUP);
  pinMode(extra_dry_pin, INPUT_PULLUP);
  pinMode(shine_plus_pin, INPUT_PULLUP);
  pinMode(hygiene_pin, INPUT_PULLUP);
  pinMode(delay_timer_pin, INPUT_PULLUP);
  pinMode(start_pin, INPUT_PULLUP);

  swSerial.begin(9600);
  RS485.begin(swSerial);

  digitalWrite(enableTransmissionPin, LOW);
}

void loop() {

  if (RS485.available()) {
    RS485.rxObj(receivedData);

    display.set(receivedData.displayText);
    display.setCustom(3, receivedData.hledstates);
    display.setCustom(4, receivedData.vledstates);
  }

  dataToSend.btnstates[PROG] = prog_btn.read();
  dataToSend.btnstates[DRY] = extradry_btn.read();
  dataToSend.btnstates[SHINE] = shineplus_btn.read();
  dataToSend.btnstates[HYGIENE] = hygiene_btn.read();
  dataToSend.btnstates[AUTOCLEAN] = autoclean_btn.read();
  dataToSend.btnstates[DELAY] = delaytimer_btn.read();
  dataToSend.btnstates[START] = start_btn.read();
  dataToSend.btnstates[CANCEL] = cancel_btn.read();

  digitalWrite(enableTransmissionPin, HIGH);
  RS485.sendDatum(dataToSend);

  display.show(); // show stored "GO" while in loop
}`

PowerBroker2, 2024-04-22:
You might have better luck having a delay in the main loop of the slave code. Beyond that, I would say strip down your code as much as possible and test with ever slightly more complexity until things break. Then debug from there.

nomakewan, 2025-04-25:
Your display board sets the transmission pin to high but never sets it to low.

Your master board sets the transmission pin to high before delaying for 50ms, meaning it only ever goes low for a few microseconds each time through the loop.

I'm pretty sure that's where your error is.