PowerBroker2/SerialTransfer #116: jpg photo Serial transmit (again!!!) doesn't work

philippedc, 2025-01-17:
Hi, many thanks for this library and your contribution. There was several threads about transmitting a picture,... I'm trying to get it working with the last library version. 
The project is building a camera door bell. 
The camera is a ESP32-CAM, the receiver is a Cheap Yellow Display ESP32.
Here is the code for the sender:
`
/*
ESP32-CAM takes photos to save on SPIFFS, and with serial transfert / RS485 wire transfer

From an idea of: http://electroniqueamateur.blogspot.com/2020/07/esp32-cam-gestion-distance-de-la-carte.html
                 https://gist.github.com/ypelletier/e47f4df32745fbdd7bc6799a42979ba3
RS485 exemple: https://microcontrollerslab.com/rs485-serial-communication-esp32-esp8266-tutorial/

about file transfer: https://github.com/PowerBroker2/SerialTransfer/tree/master
discussion: https://forum.arduino.cc/t/transfer-pictures-from-esp32-cam-to-esp32-via-serial/647488
!!! file transfer v3.1.3 only
IDE version 1.8.16 with Expressif Sytems (ESP32) version 2.0.3 
*/

//#define SAVE2SPIFFS         // to save photo to SPIFFS

// ===========================
// Enter your WiFi credentials
// ===========================
const char* ssid = "Livebox-18F6";
const char* password = "oZLaZeQHNmU4HC5hZY";

// wifi
//-------
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
WebServer server(80);

// SDCard
//---------
#include <SPIFFS.h>
#include <FS.h>

// to keep data
//-------------
#include <Preferences.h>  // https://randomnerdtutorials.com/esp32-save-data-permanently-preferences/
Preferences pref;

// camera 
//----------------
#include "esp_camera.h"
#include "Arduino.h"
#include "soc/soc.h"            // Disable brownour problems
#include "soc/rtc_cntl_reg.h"   // Disable brownour problems

// file transfer
//----------------
#include "SerialTransfer.h"     // https://github.com/PowerBroker2/SerialTransfer/tree/master
SerialTransfer myTransfer;
HardwareSerial Comm(1);

// RS485
//----------------
// note: 2nd Serial with Tx = 15, Rx = 14 to MAX485 Rx & Tx pins
#define REpin            15       // MAX485 RE/DE pins 2 & 3
#define TXpin            14       // MAX485 RO pin 4
#define RXpin            13       // MAX485 DI pin 1
#define RS485Transmit  HIGH
#define RS485Receive    LOW
#define BAUD_RATE    200000  //962100

#define flashPin          4       // white flash led
#define ledPin           33       // red led
int fileNumber = 0;               // jpeg file name index number

[... 264 lines cut ...]
  if(LoadFromSPIFFS(server.uri())) return;
  String message = "Action imprevue\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for(uint8_t i = 0; i < server.args(); i++) {
    message += " NAME:" + server.argName(i) + "\n VALUE:" + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  Serial.print(message);
}      // end of handleNotFound()

void returnOK() {
  server.send(200, "text/plain", "");
}

void returnFail(String msg) {
  server.send(500, "text/plain", msg + "\r\n");
}

`

philippedc, 2025-01-17:
Here is the receiver:

`
/*
RS485 Serial transmit from ESP32-CAM to ESP32-CYD

about serial2 for CYD: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/discussions/75
about MAX485 to MAX485: https://microcontrollerslab.com/rs485-serial-communication-esp32-esp8266-tutorial/

the RGB led pins are used for RS485 connexion:
17 TX2, 16 RX2, 4 RE

*/

// parameters
//--------------
const String VERSION = "v1.0";    // code version
#define DEBUG      1              // allow console messages

// ===========================
// Enter your WiFi credentials
// ===========================
const char* ssid = "Livebox-18F6";
const char* password = "oZLaZeQHNmU4HC5hZY";

// wifi
#include <WiFi.h>
#include <WebServer.h>
WebServer server(80);

// TFT display
//----------------
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
#include <TJpg_Decoder.h>         // https://github.com/Bodmer/TJpg_Decoder

//#include <LovyanGFX.hpp>
//#include "lgfx_ESP32_2432S028.h"

byte chosenTextFont  = 0;         // 0 or 2 or 4
const byte textFont[]  = { 4, 1, 2, 2, 1, 2 };  // 3 combinaisons font + size
#define backgroundColor  TFT_BLACK
#define textColor        TFT_WHITE
byte screenRotation  = 3;

int16_t dw = 320;
int16_t dh = 240;

// sd card part
#include "SPI.h"
#include "FS.h"
#include "SD.h"

// pinup
//-----------------
// sd card pins: default VSPI pins
#define SD_MOSI  23
#define SD_MISO  19
#define SD_SCK   18
#define SD_CS     5

// file transfer
//----------------
#include "SerialTransfer.h"     // https://github.com/PowerBroker2/SerialTransfer/tree/master
SerialTransfer myTransfer;

#define BAUD_RATE    200000
#define RS485Transmit  HIGH
#define RS485Receive    LOW
#define REpin             4
[... 244 lines cut ...]
  Serial.printf("Renaming file %s to %s\n", path1, path2);
  if (fs.rename(path1, path2)) Serial.println(F("File renamed"));
  else                         Serial.println(F("Rename failed"));
}

void DeleteFile(fs::FS &fs, const char * path) {
  Serial.printf("Deleting file: %s\n", path);
  if (fs.remove(path)) Serial.println(F("File deleted"));
  else                 Serial.println(F("Delete failed"));
}

void CopyFile(fs::FS &fs1, const char * path1, fs::FS &fs2, const char * path2) {
  Serial.printf("copying: %s to %s\n", path1, path2);
  File file1 = fs1.open(path1, FILE_READ);
  if( fs2.open(path2, FILE_READ)) Serial.printf("%s yet exists, delete it first\n", path2);
  File file2 = fs2.open(path2, FILE_WRITE);

  size_t n;  
  uint8_t buf[64];
  while((n = file1.read(buf, sizeof(buf))) > 0) {
    file2.write(buf, n);
  }
}

`

philippedc, 2025-01-17:
I cannot receive a viewable jpg photo. 
I do not understand what is wrong, it would be possible to compare the jpg file content before transmit and once receive, however I cannot configure the SD card at the same time with Serial transfer...

However if I rename the pic.jpg as pic.txt it looks like a jpg file.
Something strange :
I first transmit the image buff size: myTransfer.sendDatum(fbSize);
Once received, I do not get the same value between fbSize and the number of bytes transmitted during the for(uint16_t i=2; i<myTransfer.bytesRead; i++)
for instance on the console I can get:
  byteCount= 3416
  imageSize: 3442

I guess here is the problem

philippedc, 2025-02-01:
I get time to investigate, for instance to compare the jpeg file before serial transmit and after. 
There are very few différences between them, but there from time to time a lack of data transmitted.
Here is the file comparaison to illustrate:

[image]

philippedc, 2025-02-01:
the baud rate is 57600 for a distance of 10cm.

philippedc, 2025-02-01:
I've done more tests and it is always the same characters that are not transferred, the 2 first are those I circle in yellow on the above picture.
I've tried to downgrade V3.1.3 to 3.0.6 (older versions do not compile with expressif 2.0.3) but it doesn't change the issue

philippedc, 2025-02-02:
I'm sorry to write so much messages, but I really would like to solve this issue.

I decide to base64 code the data before transfer, and decode it once receive. There are still missing characters, but now we know what:

each trame bytes #253 (and as 2 characters are missing I suppose #254 as well) are cut off.

Never the less my code is directly coming from file transfer exemple, I use back the file transfer exemples and find the same issue:

[image]

PowerBroker2, 2025-02-02:
Thanks for bringing this to my attention - there was a slight bug in the example TX sketch. This should work instead:

```C++
#include "SerialTransfer.h"

SerialTransfer myTransfer;

const int fileSize = 2000;
char file[fileSize] = "Lorem ipsum dolor sit amet, consectetuer adipiscing elit. Aenean commodo ligula eget dolor. Aenean massa. Cum sociis natoque penatibus et magnis dis parturient montes, nascetur ridiculus mus. Donec quam felis, ultricies nec, pellentesque eu, pretium quis, sem. Nulla consequat massa quis enim. Donec pede justo, fringilla vel, aliquet nec, vulputate eget, arcu. In enim justo, rhoncus ut, imperdiet a, venenatis vitae, justo. Nullam dictum felis eu pede mollis pretium. Integer tincidunt. Cras dapibus. Vivamus elementum semper nisi. Aenean vulputate eleifend tellus. Aenean leo ligula, porttitor eu, consequat vitae, eleifend ac, enim. Aliquam lorem ante, dapibus in, viverra quis, feugiat a, tellus. Phasellus viverra nulla ut metus varius laoreet. Quisque rutrum. Aenean imperdiet. Etiam ultricies nisi vel augue. Curabitur ullamcorper ultricies nisi. Nam eget dui. Etiam rhoncus. Maecenas tempus, tellus eget condimentum rhoncus, sem quam semper libero, sit amet adipiscing sem neque sed ipsum. Nam quam nunc, blandit vel, luctus pulvinar, hendrerit id, lorem. Maecenas nec odio et ante tincidunt tempus. Donec vitae sapien ut libero venenatis faucibus. Nullam quis ante. Etiam sit amet orci eget eros faucibus tincidunt. Duis leo. Sed fringilla mauris sit amet nibh. Donec sodales sagittis magna. Sed consequat, leo eget bibendum sodales, augue velit cursus nunc, quis gravida magna mi a libero. Fusce vulputate eleifend sapien. Vestibulum purus quam, scelerisque ut, mollis sed, nonummy id, metus. Nullam accumsan lorem in dui. Cras ultricies mi eu turpis hendrerit fringilla. Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia Curae; In ac dui quis mi consectetuer lacinia. Nam pretium turpis et arcu. Duis arcu tortor, suscipit eget, imperdiet nec, imperdiet iaculis, ipsum. Sed aliquam ultrices mauris. Integer ante arcu, accumsan a, consectetuer eget, posuere ut, mauris. Praesent adipiscing. Phasellus ullamcorper ipsum rutrum nunc. Nunc nonummy metus. Vestib";
char fileName[] = "test.txt";

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  
  myTransfer.begin(Serial);
}

void loop()
{
  myTransfer.sendDatum(fileName); // Send filename
  
  uint16_t numPackets = fileSize / (MAX_PACKET_SIZE - 2); // Reserve two bytes for current file index
  
  if (fileSize % MAX_PACKET_SIZE) // Add an extra transmission if needed
    numPackets++;
  
  for (uint16_t i=0; i<numPackets; i++) // Send all data within the file across multiple packets
  {
    uint8_t  dataLen   = MAX_PACKET_SIZE - 2;
######################################################################################
######################################################################################
### This changed: ###
    uint16_t fileIndex = i * dataLen; // Determine the current file index
######################################################################################
######################################################################################

    if ((fileIndex + (MAX_PACKET_SIZE - 2)) > fileSize) // Determine data length for the last packet if file length is not an exact multiple of MAX_PACKET_SIZE-2
      dataLen = fileSize - fileIndex;
    
    uint8_t sendSize = myTransfer.txObj(fileIndex); // Stuff the current file index
    Serial.println();
    Serial.println(sendSize);
    Serial.println(dataLen);
    sendSize = myTransfer.txObj(file[fileIndex], sendSize, dataLen); // Stuff the current file data
    
    myTransfer.sendData(sendSize, 1); // Send the current file index and data
    Serial.println();
    Serial.println(sendSize);
    Serial.println(dataLen);
    Serial.println();
    delay(100);
  }
  delay(10000);
  Serial.println();
  Serial.println();
}

```

### Side note:
I've seen many people do this when posting issues and I never understood: how is it possible to post code on an issue without formatting it, look at it after posting, and not realize the unformatted code looks horrendous? How come it never occurred to you that the code should be formatted like how I did above? Why do I need to tell everyone to format code in issues???

Sorry for the rant, but I'm always bewildered when people do this and would like to understand.

philippedc, 2025-02-02:
Many thanks for the correction.

I'm sorry, I do not understand how github works. 
I never successed how to format code, even when I use the appropriate button it does not operate as expected. And as there is no way to preview...