PowerBroker2/SerialTransfer #117: Serialtransfer with Softserial and rpi pico

sjgcornish, 2025-02-12:
I have discovered that if I use this combination I get the Error Stale Packet when the baud rate is to low for example 9600 it works correctly at 38400. This is not a problem for my application but it did take me most of the day to find this out. I would however like to no why as well as to pass this information on. I am using the earlephiphower core.

nomakewan, 2025-04-25:
Depends on the size of your datum struct. At 9600 baud it takes roughly 1ms to transfer 1 byte. SerialTransfer's encapsulation adds 6 bytes on top of your payload size. So if your payload is larger than about 30 bytes, you run a real risk of timing out since the default timeout for this library is 50ms.

You can either raise your baud rate (as you already discovered), or override the default 50ms timeout to accomodate the amount of time it will take to transfer your struct.

For example, at 9600 baud to safely transfer a struct that is 50 bytes, you should have a timeout of at least 100ms.

To override the default, use this `begin` format:

```cpp
SerialTransfer myTransfer;
Serial2.begin(9600);

myTransfer.begin(Serial2, false, Serial, 100);
```

The first parameter tells SerialTransfer what interface to use, the second enables/disables debug logging to an alternative Serial interface, the third parameter defines which Serial interface to use for debug logging (irrelevant here because debugging is set to `false`), and the last parameter is the timeout in milliseconds.

Hope that helps!

sjgcornish, 2025-04-26:
Many thanks for the information.

closed 2025-07-14 by PowerBroker2